Aarhus-University-MPE/BioDiscover #17: Species name (and likely sample name) may not end with a dot

PhilippMRehsen, 2022-06-23:
### Ice-Nucleation-Controller version

1.9.1-beta

### Area(s) with issue?

General

### Steps to reproduce

1) Entering a species name which ends with a dot (e.g. _Gammarus sp_., "sp." being commonly used in several species collections) works fine until the start of the imaging process: 
[image]
 
This is likely because the calibration pictures cannot be saved, since the folder in Windows may not end with a dot and no folder is created after taking calibration pictures. However, taking calibration pictures is working fine. 
2) Pressing the "Change Cropping" button does nothing.

### ✔️ Expected Behavior

1) A warning message / error message should appear when trying to use a species name ending with a dot (and / or other "illegal" windows characters). This should happen latest when taking calibration pictures (or after failing to create a folder including these pictures).

2) Trying to crop calibration files which cannot be found should also display the "Calibration image(s) are missing!" error.

### ❌ Actual Behavior

1) Just the "Calibration images are missing" error appears, which likely causes the user to again take calibration pictures. This however seemingly works fine, since technically the device takes these images, but they cannot be saved. 

2) (As stated above): Nothing happens when pressing "Change Cropping"

MadsR, 2022-06-24:
This is mainly due to the automatic name generation (and folder creation), which is linked to the sample name.
Similarly adding a `/` character will cause errors to appear as well.

**Potential Fix**
Omitt (or change) special characters when generating folder and file names

Sample Name input: 
`Beetle.Southern/Eastern`

Previous output: 
`images/camerasettings/speciesName/Beetle.Southern/Eastern/Calibration/images...`

 - Not a valid path - will result in error when trying to save/load images

With proposed solution: 
`images/camerasettings/speciesName/Beetle_Southern_Eastern/Calibration/images...`

PhilippMRehsen, 2022-06-24:
I think omitting special characters is a perfectly fine solution. 
However,