Aarhus-University-MPE/BioDiscover #20: Automated detection of unwanted images (including unwanted objects e.g. tweezers)

PhilippMRehsen, 2022-07-26:
### Description of the new feature / enhancement

The imaging process with BioDiscover includes dropping specimen into the cuvette, which is for now done mostly manually. Specimen can occasionally get stuck on the site of the cuvette prior to being completely submerged in ethanol. With a manual mode of operation, this can get detected fast and is likely solved by pushing down the specimen into the cuvette with forceps or tweezers. During this scenario, it frequently happens that the tip of used tweezers are imaged (and measured). [image]
Detecting intersections of the binary image with the edge of an image would also allow the detection of 

Suggested solution:
Automatically detect objects touching the upper edge of an image. This way "wrong" images wouldn't be taken or could be detected fast during post processing. E.g.: A column with something like "Object detected at edge" => "yes/no" could be used to filter images to either have a look at these images or exclude these images from further analyses. 

### Scenario when this would be used?

This features would be extremely helpful when dealing with small specimen, since they frequently get attached to the site of the cuvette.

### Supporting information

I could provide more images with forceps on them if needed.

MadsR, 2022-07-28:
This is already part of the detection algorithm. The object must follow a set of criteria before cropping and saving begins:

- area > threshold
- Crop min > image min
- Crop max < Image max

The last two check for collisions with the edge, (see manual page 30).

Times where this check fail is when an object is separated into multiple blobs, as the checks are only done on the largest object.

**Multiple large objects in frame criteria**
A previous check was made that required a maximum of 1 "large" object in frame; if more than one blob met the area > threshold criteria, no cropping was made. This would previously fail, but could most likely be reimplemented to avoid "bad images".

MadsR, 2022-07-28:
You can see the criteria described above in action, by starting image capture and lower a tweezer into the cuvette.

You should clearly see the tweezer in frame, and the object marked in red on the binary feed. But no images should be cropped and saved.

PhilippMRehsen, 2022-08-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MadsR, 2022-08-08:
I agree the best solution to these would be to reimplement the "maximum large objects" criteria.

Image 1 and 3 does however appear to fail the edge detection; given how the tweezers appear as the largest object in frame and quite clearly touch the upper edge of the image.