Aarhus-University-MPE/BioDiscover #34: Alternative Camera mount functionalities

MadsR, 2022-08-30:
### Description of the new feature / enhancement

Create functionality that allows for alternative camera orientation on non XL versions.

In practicality system uses XL version calculations with objects moving horizontal in frame, instead of the usual vertical.

### Scenario when this would be used?

For systems with upgraded camera mount (vertical/horizontal) orientation.

PhilippMRehsen, 2023-01-31:
Is there any workaround for now?
E.g. specifying the camera view (rotated / original) somewhere manually without an automatic detection?

MadsR, 2023-01-31:
There are two ways to get around this.

**1. Onboard Firmware _Hack_**
Modifying the onboard firmware serial ID to include XL, will trick the system into thinking it is an XL variant and apply the oriented calculations. The pump can still be activated by applying the system override ``CTRL+D``.

**2. Onboard Firmware and Software Update**
Modify the onboard firmware serial ID to include a new Horizontal tag and add the detection of this to the SRS software, which applies the oriented calculations and enables the pump.

Solution 1 takes 30 seconds to do, while solution 2 takes about an hour. I will send you an email where we can discuss how we proceed.