Aarhus-University-MPE/BioDiscover #48: Biodiscover version ID stored in output

LDalby, 2023-02-06:
### Description of the new feature / enhancement

To be able to distinguish the biodiscover machine (the big one or the small one). 

### Scenario when this would be used?

-

LDalby, 2024-04-30:
@kaviecos 👀

kaviecos, 2024-04-30:
We should also store the rotation of the camera

LDalby, 2024-04-30:
Alternatively we could just make sure that the roi columns are calibrated to the rotation so that e.g. `roi_top` really is the top.

kaviecos, 2024-04-30:
Maybe both? If we want to recreate the crop from the original image (or the original image from crop + calibration) we need roi_top to be the top relative to the image. And to determine fall speed we need roi_top to be the real top.