Aarhus-University-MPE/BioDiscover #64: Empty lines in Image data.csv

LDalby, 2023-05-10:
### BioDiscover Software version

?

### Area(s) with issue?

Image Saving

### Steps to reproduce

While processing image data.csv's I encounter some files with a number of empty lines. It can be from just a couple and up to several hundreds. 
The empty lines are in chunks within the file.

### ✔️ Expected Behavior

No empty lines.

LDalby, 2023-05-10:
Possibly related to #62

MadsR, 2023-05-10:
Could be related to the lack of #16 , resulting in operators manually clearing data from files.

LDalby, 2023-05-10:
Some of these have been run through the machine this spring, so there is a chance that the operators can actually remember if they've done something like that.
I will ask them and report back.